DIY-Flow-Bench/DIY-Flow-Bench #330: Review NEW ESPAsyncWebServer Library

DeeEmm, 2025-01-24:
Just noticed that there is a NEW ESPAsyncWebServer library.

https://github.com/ESP32Async/ESPAsyncWebServer

It appears that someone has picked up maintaining the old library as the old library now links to the repo above.

It states that there are LOTS of bug fixes. 👍 😄 

This library is in part responsible for some of the speed and page load issues that we have experienced. So I am keen to test. 

Would be good to do this before RC9

DeeEmm, 2025-01-25:
Updated to the new ESPAsynCWebServer library. This also included an update to the Async_TCP library as well, as they are now packaged together.

This change also highlighted a couple of areas for improvement which have also been updated. Specifically how we process static files such as javascript and css files. Previously these were decompressed before being sent to the browser. The reason for this is that this is how we need to handle files that are sent to the templating engine. Unfortunately when files were moved from SPIFFS into PROGMEM and compressed, there was no differentiation made between files that require template vars to be processed (file decompressed) and those that don't.

So now we just send the static files directly to the browser without touching them at all, which is not just faster but also less memory intensive. To achieve this the compression format needed to be changed from zlib to gzip. This change alone should be a big win in terms of memory and page load speed.

The update is now in the DEV branch

closed 2025-01-25 by DeeEmm

stefan63, 2025-01-25:
Must try the new DEV tonight :0)