DIY-Flow-Bench/DIY-Flow-Bench #363: [BUG]

ModchipMan, 2025-02-10:
**Software Version**
- RELEASE "V2.0-RC8"
- BUILD_NUMBER "2502060002"

NOTE: Both can be found in version.json or by pressing the 'Info' button on the configuration page in the GUI

**Describe the bug**
GUI wont load, I can connect to DIYFB, but diyfb.local will not load,

**Expected behavior**
To get to DIYFB GUI

**Steps to reproduce the bug**
Outline the steps to reproduce the behavior:
1. Load firmware with esp tool or firmware flasher
2. Connect to DIYFB
3. try and open diyfb.local web browser "microsoft edge" and see page not loading

**ESP32 (please include additional information as required):**
 - Hardware Type  Wemos D1 R32
 - ESP Firmware Version (not known)
 - 
**Additional context**
Add any other context about the problem here.

DeeEmm, 2025-02-10:
Can you connect using the IP address shown in the serial monitor?

ModchipMan, 2025-02-11:
checked serial monitor, found it is rebooting after it cant find BME280, which i dont have installed..
I do have a BME680.
I do have a BME280 but its 3.3v and could be a BMP280 copy..
I will probably just wait until the code supports BME680..
Iam also waiting on the shift leveller that I ordered about a month ago, and then that completes my V2.0 shield.
ohh I do have pink board ESP32-S3 ready to go when the code gets there, as well as my Wemos